phifogg/ioBroker.sainlogic #174: Scheduler received data for unkown command XX subcommand YY

Klaus-Lucas, 2023-07-24:
Hallo,
nutze deinen Adapter mit einer DP1500 Wetterstation in folgernder Umgebung:
admin: 6.6.1
Node.js: v18.16.1
NPM: v9.5.1
js-controller: 5.0.8

Nachdem der Adapter gestartet wurde, kommt obige Meldung nach "Scheduler connected to weather station":

`2023-07-24 21:58:29.198 - info: sainlogic.0 (5416) starting. Version 0.10.5 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v18.16.1, js-controller: 5.0.8
2023-07-24 21:58:29.243 - info: sainlogic.0 (5416) Starting Scheduler
2023-07-24 21:58:29.245 - info: sainlogic.0 (5416) Scheduler call ws_getfirmware activated
2023-07-24 21:58:29.246 - info: sainlogic.0 (5416) Scheduler call ws_getcurrent activated
2023-07-24 21:58:29.247 - debug: sainlogic.0 (5416) Weatherstation IP: 192.168.178.45
2023-07-24 21:58:29.248 - debug: sainlogic.0 (5416) Weatherstation port: 45000
2023-07-24 21:58:29.249 - debug: sainlogic.0 (5416) Scheduler Interval: 5
2023-07-24 21:58:29.250 - debug: sainlogic.0 (5416) Listener IP: 192.168.178.41
2023-07-24 21:58:29.251 - debug: sainlogic.0 (5416) Listener port: 45000
2023-07-24 21:58:29.251 - debug: sainlogic.0 (5416) Listener path: /weatherstation/updateweatherstation.php
2023-07-24 21:58:29.252 - debug: sainlogic.0 (5416) Listener protocol: ecowitt
2023-07-24 21:58:29.253 - debug: sainlogic.0 (5416) Forward URL:
2023-07-24 21:58:29.254 - info: sainlogic.0 (5416) Starting Listener
2023-07-24 21:58:34.250 - info: sainlogic.0 (5416) Scheduler pulling for new data
2023-07-24 21:58:34.420 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,80,3,83
2023-07-24 21:58:34.435 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff50120e475731303030415f56312e372e36bf
2023-07-24 21:58:34.438 - debug: sainlogic.0 (5416) Data Command received: 80 subcommand 71
2023-07-24 21:58:34.439 - error: sainlogic.0 (5416) Scheduler received data for unkown command 80 subcommand 71
2023-07-24 21:58:34.439 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,11,0,6,4,4,25
2023-07-24 21:58:34.448 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff5b040261
2023-07-24 21:58:34.449 - debug: sainlogic.0 (5416) Data Command received: 91 subcommand 97
2023-07-24 21:58:34.449 - error: sainlogic.0 (5416) Scheduler received data for unkown command 91 subcommand 97
2023-07-24 21:58:34.451 - debug: sainlogic.0 (5416) FW Scheduler Connection closed
2023-07-24 21:58:39.251 - info: sainlogic.0 (5416) Scheduler pulling for new data
2023-07-24 21:58:39.257 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,80,3,83
2023-07-24 21:58:39.274 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff50120e475731303030415f56312e372e36bf
2023-07-24 21:58:39.274 - debug: sainlogic.0 (5416) Data Command received: 80 subcommand 71
2023-07-24 21:58:39.275 - error: sainlogic.0 (5416) Scheduler received data for unkown command 80 subcommand 71
2023-07-24 21:58:39.275 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,11,0,6,4,4,25
2023-07-24 21:58:39.284 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff5b040261
2023-07-24 21:58:39.285 - debug: sainlogic.0 (5416) Data Command received: 91 subcommand 97
2023-07-24 21:58:39.286 - error: sainlogic.0 (5416) Scheduler received data for unkown command 91 subcommand 97
2023-07-24 21:58:39.289 - debug: sainlogic.0 (5416) FW Scheduler Connection closed
2023-07-24 21:58:44.251 - info: sainlogic.0 (5416) Scheduler pulling for new data
2023-07-24 21:58:44.257 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,80,3,83
2023-07-24 21:58:44.274 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff50120e475731303030415f56312e372e36bf
2023-07-24 21:58:44.275 - debug: sainlogic.0 (5416) Data Command received: 80 subcommand 71
2023-07-24 21:58:44.275 - error: sainlogic.0 (5416) Scheduler received data for unkown command 80 subcommand 71
2023-07-24 21:58:44.276 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,11,0,6,4,4,25
2023-07-24 21:58:44.285 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff5b040261
2023-07-24 21:58:44.286 - debug: sainlogic.0 (5416) Data Command received: 91 subcommand 97
2023-07-24 21:58:44.286 - error: sainlogic.0 (5416) Scheduler received data for unkown command 91 subcommand 97
2023-07-24 21:58:44.289 - debug: sainlogic.0 (5416) FW Scheduler Connection closed
2023-07-24 21:58:49.251 - info: sainlogic.0 (5416) Scheduler pulling for new data
2023-07-24 21:58:49.264 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,80,3,83
2023-07-24 21:58:49.284 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff50120e475731303030415f56312e372e36bf
2023-07-24 21:58:49.285 - debug: sainlogic.0 (5416) Data Command received: 80 subcommand 71
2023-07-24 21:58:49.285 - error: sainlogic.0 (5416) Scheduler received data for unkown command 80 subcommand 71
2023-07-24 21:58:49.286 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,11,0,6,4,4,25
2023-07-24 21:58:49.295 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff5b040261
2023-07-24 21:58:49.296 - debug: sainlogic.0 (5416) Data Command received: 91 subcommand 97
2023-07-24 21:58:49.296 - error: sainlogic.0 (5416) Scheduler received data for unkown command 91 subcommand 97
2023-07-24 21:58:49.299 - debug: sainlogic.0 (5416) FW Scheduler Connection closed
2023-07-24 21:58:54.251 - info: sainlogic.0 (5416) Scheduler pulling for new data
2023-07-24 21:58:54.257 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,80,3,83
2023-07-24 21:58:54.274 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff50120e475731303030415f56312e372e36bf
2023-07-24 21:58:54.275 - debug: sainlogic.0 (5416) Data Command received: 80 subcommand 71
2023-07-24 21:58:54.275 - error: sainlogic.0 (5416) Scheduler received data for unkown command 80 subcommand 71
2023-07-24 21:58:54.276 - debug: sainlogic.0 (5416) Scheduler connected to weather station run 255,255,11,0,6,4,4,25
2023-07-24 21:58:54.286 - debug: sainlogic.0 (5416) FW Scheduler Received data string: ffff5b040261
2023-07-24 21:58:54.287 - debug: sainlogic.0 (5416) Data Command received: 91 subcommand 97
2023-07-24 21:58:54.288 - error: sainlogic.0 (5416) Scheduler received data for unkown command 91 subcommand 97
2023-07-24 21:58:54.290 - debug: sainlogic.0 (5416) FW Scheduler Connection closed
2023-07-24 21:58:54.958 - info: host.iobroker "system.adapter.sainlogic.0" disabled
2023-07-24 21:58:54.960 - info: host.iobroker stopInstance system.adapter.sainlogic.0 (force=false, process=true)
2023-07-24 21:58:54.974 - info: sainlogic.0 (5416) Got terminate signal TERMINATE_YOURSELF
2023-07-24 21:58:54.976 - info: sainlogic.0 (5416) Sainlogic Adapter gracefully shut down...
2023-07-24 21:58:54.977 - info: sainlogic.0 (5416) terminating
2023-07-24 21:58:54.978 - info: sainlogic.0 (5416) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2023-07-24 21:58:54.979 - info: sainlogic.0 (5416) Listener closed.
2023-07-24 21:58:55.235 - info: host.iobroker stopInstance system.adapter.sainlogic.0 send kill signal
2023-07-24 21:58:55.480 - info: sainlogic.0 (5416) terminating
2023-07-24 21:58:55.713 - info: host.iobroker instance system.adapter.sainlogic.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)`

Daten werden in der WSView-Plus App angezeigt.
Hab ich irgendwo was verbastelt?

phifogg, 2023-07-25:
Nein, denke du hast das richtig gemacht. Die Station reagiert nur auf die Komandos anders als ich es erwarte.
Leider habe ich keine DP1500 zum Testen und bis jetzt konnte mir keiner sagen wie genau die funktioniert.

Wenn es dazu irgendwo eine Anleitung gibt kann ich den Scheduler noch einpflegen. Bis dahin musst Du wohl mit dem Listener Modus agieren.

Klaus-Lucas, 2023-07-25:
Sehe, dass ich einen DP1000_Pro habe. Würde es dir helfen, wenn ich ihn dir zuschicken?
"Bis dahin musst Du wohl mit dem Listener Modus agieren." OK, das läuft

phifogg, 2023-07-25:
Das ist doch schonmal gut.
Nein, das Gerät zu haben hilft nur bedingt. Ich bräuchte infos wie ich die Daten von der Station zerlegen muss. Da kommt eben sowas wie

`ffff50120e475731303030415f56312e372e36bf`

Das sieht ja ähnlich aus wie das was meine Station liefert. Aber eben nicht gleich. Vermutlich ist es nicht schwierig, wenn man weiss wie. Für die ELV gibt es eben ne schöne Doku.

phifogg, 2023-07-25:
Wenn Du mal mutig sein willst :D
Im Order Lib gibt es eine Datei constants.js, hier müssten ein paar Zeilen getauscht werden, und dann sollte ich zumindest die Softwareversion auslesen können. Hab ich gerade mal mit dem Log von Dir probiert-

Etwa auf Zeile 1579 wir ein Array COMMANS definiert. Schau da mal rein, es gibt ein block 'Get Firmware'. Hier muss das subcommand_init und die length geändert werden. Es sollte so aussehen:

`    {
        name: 'Get Firmware (DP1000)(',
        configurable: true,
        config_variable: 'ws_getfirmware',
        channel: 'info',
        command: [0xFF, 0xFF, 0x50, 0x03, 0x53],
        command_int: 80,
        subcommand_int: 71,
        parser: new BinaryParser()
            .endianess('big').seek(5)
            .string('softwaretype', {
                encoding: 'utf8',
                length: 14
            })
    },`

Dann aktiviere mal im Zeitplan nur die 'Firmware abrufen' und starte den Scheduler. Vielleicht können wir uns ja so der Sache nähern...

Klaus-Lucas, 2023-07-25:
Klar, mach ich. Parallel habe ich mal bei ecowitt wegen Doku. nachgefragt. 

https://github.com/phifogg/ioBroker.sainlogic/issues/174#issuecomment-1649733123

Klaus-Lucas, 2023-07-25:
da bist du wohl auf dem richtigem Wege:

2023-07-25 16:58:48.510 - info: sainlogic.0 (8516) starting. Version 0.10.5 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v18.17.0, js-controller: 5.0.8

2023-07-25 16:58:48.555 - info: sainlogic.0 (8516) Starting Scheduler

2023-07-25 16:58:48.556 - info: sainlogic.0 (8516) Scheduler call ws_getfirmware activated

2023-07-25 16:58:48.557 - debug: sainlogic.0 (8516) Weatherstation IP: 192.168.178.45

2023-07-25 16:58:48.558 - debug: sainlogic.0 (8516) Weatherstation port: 45000

2023-07-25 16:58:48.559 - debug: sainlogic.0 (8516) Scheduler Interval: 15

2023-07-25 16:59:03.560 - info: sainlogic.0 (8516) Scheduler pulling for new data

2023-07-25 16:59:03.656 - debug: sainlogic.0 (8516) Scheduler connected to weather station run 255,255,80,3,83

2023-07-25 16:59:03.676 - debug: sainlogic.0 (8516) FW Scheduler Received data string: ffff50120e475731303030415f56312e372e36bf

2023-07-25 16:59:03.679 - debug: sainlogic.0 (8516) Data Command received: 80 subcommand 71

2023-07-25 16:59:03.682 - debug: sainlogic.0 (8516) Data object: {"softwaretype":"GW1000A_V1.7.6"}

2023-07-25 16:59:03.683 - info: sainlogic.0 (8516) Scheduler updating IOBroker states

2023-07-25 16:59:03.686 - debug: sainlogic.0 (8516) Setting value from data for info.softwaretype to GW1000A_V1.7.6

2023-07-25 16:59:03.689 - debug: sainlogic.0 (8516) FW Scheduler Connection closed

2023-07-25 16:59:18.559 - info: sainlogic.0 (8516) Scheduler pulling for new data

2023-07-25 16:59:18.567 - debug: sainlogic.0 (8516) Scheduler connected to weather station run 255,255,80,3,83

2023-07-25 16:59:18.586 - debug: sainlogic.0 (8516) FW Scheduler Received data string: ffff50120e475731303030415f56312e372e36bf

2023-07-25 16:59:18.587 - debug: sainlogic.0 (8516) Data Command received: 80 subcommand 71

2023-07-25 16:59:18.587 - debug: sainlogic.0 (8516) Data object: {"softwaretype":"GW1000A_V1.7.6"}

2023-07-25 16:59:18.588 - info: sainlogic.0 (8516) Scheduler updating IOBroker states

2023-07-25 16:59:18.590 - debug: sainlogic.0 (8516) Setting value from data for info.softwaretype to GW1000A_V1.7.6

2023-07-25 16:59:18.598 - debug: sainlogic.0 (8516) FW Scheduler Connection closed

 

Wird aber nicht in Objekte/Model angezeigt/übernommen.

Klaus-Lucas, 2023-07-25:
Wunderground Listener klappte auch nicht:

2023-07-25 17:22:29.911 - info: sainlogic.0 (25387) starting. Version 0.10.5 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v18.17.0, js-controller: 5.0.8
2023-07-25 17:22:29.960 - debug: sainlogic.0 (25387) Listener IP: 192.168.178.41
2023-07-25 17:22:29.961 - debug: sainlogic.0 (25387) Listener port: 45000
2023-07-25 17:22:29.962 - debug: sainlogic.0 (25387) Listener path: /weatherstation/updateweatherstation.php?
2023-07-25 17:22:29.963 - debug: sainlogic.0 (25387) Listener protocol: wunderground
2023-07-25 17:22:29.963 - debug: sainlogic.0 (25387) Forward URL:
2023-07-25 17:22:29.965 - info: sainlogic.0 (25387) Starting Listener

2023-07-25 17:23:14.849 - warn: sainlogic.0 (25387) Listener received illegal request: /weatherstation/updateweatherstation.php?ID=IMOERS179&PASSWORD=UNcqivm3&tempf=-9999&humidity=-9999&dewptf=-9999&windchillf=-9999&winddir=-9999&windspeedmph=-9999&windgustmph=-9999&rainin=0.000&dailyrainin=0.000&weeklyrainin=0.445&monthlyrainin=1.535&yearlyrainin=16.815&solarradiation=-9999&UV=-9999&indoortempf=-9999&indoorhumidity=-9999&baromin=29.790&soilmoisture=42&soilmoisture2=40&soilmoisture3=35&soilmoisture4=31&dateutc=now&softwaretype=GW1000A_V1.7.6&action=updateraw&realtime=1&rtfreq=5

2023-07-25 17:24:15.830 - warn: sainlogic.0 (25387) Listener received illegal request: /weatherstation/updateweatherstation.php?ID=IMOERS179&PASSWORD=UNcqivm3&tempf=-9999&humidity=-9999&dewptf=-9999&windchillf=-9999&winddir=-9999&windspeedmph=-9999&windgustmph=-9999&rainin=0.000&dailyrainin=0.000&weeklyrainin=0.445&monthlyrainin=1.535&yearlyrainin=16.815&solarradiation=-9999&UV=-9999&indoortempf=-9999&indoorhumidity=-9999&baromin=29.793&soilmoisture=42&soilmoisture2=40&soilmoisture3=35&soilmoisture4=31&dateutc=now&softwaretype=GW1000A_V1.7.6&action=updateraw&realtime=1&rtfreq=5

[image]

Klaus-Lucas, 2023-07-25:
> Ja, das WU-Protokoll geht irgendwie nicht bei der Station. Das hatte ich schonmal mit jemand. Haben wir nicht rausbekommen warum. Sorry. Könnte natürlich an noch mehr von den schrägen -9999 werten liegen. Ich kann mal was einbauen um die abzufangen.
> 

Die schrägen Werte liegen ggf. daran, dass das Hypersonic-Aerometer momentan wegen Defekt nicht online ist, aber definiert ist.

> Gut dass der Scheduler mal den Versionsstring rauskriegt. Das ist doch mal was. Er sagt doch 'Setting value from data for info.softwaretype to GW1000A_V1.7.6'
> 
> Kommt das nicht an?

Schrieb oben schon: Wird aber nicht in Objekte/Model angezeigt/übernommen. 
Getestet mit iobroker1 (Beta)

> 
> Für den Listener würde ich empfehlen das '?' in ioB zu entfernen. Dann probiere mal in der Konfiguration der Station mit und ohne.

Habe den Pfad auf "/weatherstation/updateweatherstation.php"  und Protokoll "Wunderground" gesetzt:

Listener received illegal request: /weatherstation/updateweatherstation.phpID=IMOERS179&PASSWORD=UNcqivm3&tempf=-9999&humidity=-9999&dewptf=-9999&windchillf=-9999&winddir=-9999&windspeedmph=-9999&windgustmph=-9999&rainin=0.000&dailyrainin=0.000&weeklyrainin=0.445&monthlyrainin=1.535&yearlyrainin=16.815&solarradiation=-9999&UV=-9999&indoortempf=-9999&indoorhumidity=-9999&baromin=29.802&soilmoisture=42&soilmoisture2=39&soilmoisture3=35&soilmoisture4=31&dateutc=now&softwaretype=GW1000A_V1.7.6&action=updateraw&realtime=1&rtfreq=5

Klaus-Lucas, 2023-07-26:
Ist die Dokumentation hier vielleicht hilfreich:
https://github.com/merbanan/rtl_433/blob/master/src/devices/fineoffset_wh1050.c
Fine Offset ist der Original-Hersteller der vielen Derivate vertrieben von Ecowitt, Froggit....

phifogg, 2023-07-26:
Das Repo ist mal super spannend, der hat ja für einige Stationen was gefunden. Muss ich mir genauer anschauen.

Mit deinen beiden IOB's hast mich etwas verwirrt. Versteh ich richtig, der stable (2) geht alles, mit dem (1) nicht?
Und was ist bei 1 nun installiert?

Klaus-Lucas, 2023-07-27:
Kennst du übrigens diese Seite: [https://github.com/SBorg2014/WLAN-Wetterstation](https://github.com/SBorg2014/WLAN-Wetterstation)

phifogg, 2023-07-27:
Ja, das SBorg Ding kenne ich, ist ja quasi 'nur' der Listenerteil aus meinem Adapter - oder umgekehrt. Wie Du magst, das Skript gab es zuerst.

Einen Forwarder habe ich auch drin, wobei ich 'einfach' die Daten genau gleich wie ich sie krieg weiterreiche.

Klaus-Lucas, 2023-07-27:
Das foshkplugin ist da aber bei den Weiterleitungen nicht begrenzt. Man kann also:
- deinen Adapter 
- Private Weather Tablet
- ....
versorgen.

phifogg, 2023-08-26:
Dann müsstest du einfach als weiterleitungs-URL das hier eintragen:

http://192.168.178.17:8572/

Das müsste reichen.

Klaus-Lucas, 2023-08-26:
Ok, danke für die Hilfe

closed 2023-08-26 by Klaus-Lucas