phifogg/ioBroker.sainlogic #180: Leaf-wetness sensor Froggit DP10 / Ecovit WN35 not supported?

Petergoge, 2023-08-09:
I'm using DP1500 and I'm getting values for
sainlogic.0.weather.current.indoorhumidity
sainlogic.0.weather.current.indoortemp
sainlogic.0.weather.current.pressurerel

But I'm missing the Leaf-wetness sensor Froggit DP10 / Ecovit WN35. 

Can I help you importing it?
In WSView pro its called _Multi_CH Leaf_  ID: 2C01

btw.: the readme said that it has to be set to ecovit, but I just getting values from the DP1500 by using wunderground with /folder/updateweatherstation.php?

Best regards
Daniel

phifogg, 2023-08-09:
sure, can you put the adapter in debug mode and get me the output of one run? This should list what the sensor is reporting...

Petergoge, 2023-08-10:
`<html>
<body>

sainlogic.0 | 2023-08-10 16:53:07.913 | warn | State value to set for "sainlogic.0.weather.current.uvi" has value "-9999" less than min "0"
-- | -- | -- | --
sainlogic.0 | 2023-08-10 16:53:07.912 | warn | State value to set for "sainlogic.0.weather.current.solarradiation" has value "-1266873" less than min "0"
sainlogic.0 | 2023-08-10 16:53:07.912 | warn | State value to set for "sainlogic.0.weather.current.windgustspeed" has value "-16091.8" less than min "0"
sainlogic.0 | 2023-08-10 16:53:07.911 | warn | State value to set for "sainlogic.0.weather.current.windspeed" has value "-16091.8" less than min "0"
sainlogic.0 | 2023-08-10 16:53:07.910 | warn | State value to set for "sainlogic.0.weather.current.winddir" has value "-9999" less than min "0"
sainlogic.0 | 2023-08-10 16:53:07.910 | warn | State value to set for "sainlogic.0.weather.current.windchilltemp" has value "-5572.8" less than min "-40"
sainlogic.0 | 2023-08-10 16:53:07.909 | warn | State value to set for "sainlogic.0.weather.current.dewpointtemp" has value "-5572.8" less than min "-40"
sainlogic.0 | 2023-08-10 16:53:07.908 | warn | State value to set for "sainlogic.0.weather.current.outdoorhumidity" has value "-9999" less than min "0"
sainlogic.0 | 2023-08-10 16:53:07.867 | warn | State value to set for "sainlogic.0.weather.current.outdoortemp" has value "-5572.8" less than min "-40"
sainlogic.0 | 2023-08-10 16:53:07.794 | debug | Setting value from data for info.softwaretype to GW1100A_V2.2.5
sainlogic.0 | 2023-08-10 16:53:07.794 | debug | Target for pressurerel unit is set: hPa, using conversion: null
sainlogic.0 | 2023-08-10 16:53:07.793 | debug | Setting value from data for weather.current.pressurerel to 1020.4
sainlogic.0 | 2023-08-10 16:53:07.793 | debug | Setting value from data for weather.current.indoorhumidity to 61
sainlogic.0 | 2023-08-10 16:53:07.793 | debug | Target for indoortemp unit is set: °C, using conversion: null
sainlogic.0 | 2023-08-10 16:53:07.792 | debug | Setting value from data for weather.current.indoortemp to 23.4
sainlogic.0 | 2023-08-10 16:53:07.792 | debug | Setting value from data for weather.current.uvi to -9999
sainlogic.0 | 2023-08-10 16:53:07.792 | debug | Target for solarradiation unit is set: lx, using conversion: null
sainlogic.0 | 2023-08-10 16:53:07.791 | debug | Setting value from data for weather.current.solarradiation to -1266873
sainlogic.0 | 2023-08-10 16:53:07.791 | debug | Target for windgustspeed unit is set: km/h, using conversion: null
sainlogic.0 | 2023-08-10 16:53:07.791 | debug | Setting value from data for weather.current.windgustspeed to -16091.8
sainlogic.0 | 2023-08-10 16:53:07.790 | debug | Target for windspeed unit is set: km/h, using conversion: null
sainlogic.0 | 2023-08-10 16:53:07.790 | debug | Setting value from data for weather.current.windspeed to -16091.8
sainlogic.0 | 2023-08-10 16:53:07.790 | error | Wind heading could not be determined: wind direction outside boundary
sainlogic.0 | 2023-08-10 16:53:07.789 | debug | Determining wind heading for -9999 and precision 16
sainlogic.0 | 2023-08-10 16:53:07.788 | debug | Setting value from data for weather.current.winddir to -9999
sainlogic.0 | 2023-08-10 16:53:07.788 | debug | Target for windchilltemp unit is set: °C, using conversion: null
sainlogic.0 | 2023-08-10 16:53:07.788 | debug | Setting value from data for weather.current.windchilltemp to -5572.8
sainlogic.0 | 2023-08-10 16:53:07.787 | debug | Target for dewpointtemp unit is set: °C, using conversion: null
sainlogic.0 | 2023-08-10 16:53:07.787 | debug | Setting value from data for weather.current.dewpointtemp to -5572.8
sainlogic.0 | 2023-08-10 16:53:07.786 | debug | Setting value from data for weather.current.outdoorhumidity to -9999
sainlogic.0 | 2023-08-10 16:53:07.785 | debug | Target for outdoortemp unit is set: °C, using conversion: null
sainlogic.0 | 2023-08-10 16:53:07.784 | debug | Setting value from data for weather.current.outdoortemp to -5572.8
sainlogic.0 | 2023-08-10 16:53:07.748 | debug | No mapping found for dataelement: last_listener_update
sainlogic.0 | 2023-08-10 16:53:07.748 | debug | No mapping found for dataelement: rtfreq
sainlogic.0 | 2023-08-10 16:53:07.747 | debug | No mapping found for dataelement: realtime
sainlogic.0 | 2023-08-10 16:53:07.747 | debug | No mapping found for dataelement: action
sainlogic.0 | 2023-08-10 16:53:07.746 | debug | Extracting value for info.softwaretype(softwaretype)
sainlogic.0 | 2023-08-10 16:53:07.746 | debug | No mapping found for dataelement: dateutc
sainlogic.0 | 2023-08-10 16:53:07.745 | debug | No mapping found for dataelement: leafwetness
sainlogic.0 | 2023-08-10 16:53:07.745 | debug | Extracting value for weather.current.pressurerel(baromin)
sainlogic.0 | 2023-08-10 16:53:07.744 | debug | Extracting value for weather.current.indoorhumidity(indoorhumidity)
sainlogic.0 | 2023-08-10 16:53:07.743 | debug | Extracting value for weather.current.indoortemp(indoortempf)
sainlogic.0 | 2023-08-10 16:53:07.743 | debug | Extracting value for weather.current.uvi(UV)
sainlogic.0 | 2023-08-10 16:53:07.742 | debug | Extracting value for weather.current.solarradiation(solarradiation)
sainlogic.0 | 2023-08-10 16:53:07.741 | debug | Extracting value for weather.current.windgustspeed(windgustmph)
sainlogic.0 | 2023-08-10 16:53:07.740 | debug | Extracting value for weather.current.windspeed(windspeedmph)
[... cut ...]
sainlogic.0 | 2023-08-10 16:52:48.993 | debug | Listener protocol: wunderground
sainlogic.0 | 2023-08-10 16:52:48.992 | debug | Listener path: /wetterstation/updateweatherstation.php
sainlogic.0 | 2023-08-10 16:52:48.992 | debug | Listener port: 49160
sainlogic.0 | 2023-08-10 16:52:48.991 | debug | Listener IP: 0.0.0.0
sainlogic.0 | 2023-08-10 16:52:48.960 | info | starting. Version 0.11.1 (non-npm: phifogg/ioBroker.sainlogic) in /opt/iobroker/node_modules/iobroker.sainlogic, node: v14.18.3, js-controller: 4.0.24

</body>
</html>`

phifogg, 2023-08-10:
Good, this is what I need. Like leafwetness is one you want. I can add this to the next version.
There is another thing i notice, quite some values are pretty awkward, like -9999. Maybe you want to check that. I need to add some error handling for this as well.

Petergoge, 2023-08-11:
Just for safety I reinstalled the whole adapter.
At the first run there are just a few lower warnings:
<html>
<body>

sainlogic.0 | 2023-08-11 15:29:21.959 | warn | This object will not be created in future versions. Please report this to the developer.
-- | -- | -- | --
sainlogic.0 | 2023-08-11 15:29:21.958 | warn | Object sainlogic.0.weather.current.pressureabs is invalid: Default value has to be type "number" but received type "string"
sainlogic.0 | 2023-08-11 15:29:21.958 | warn | This object will not be created in future versions. Please report this to the developer.
sainlogic.0 | 2023-08-11 15:29:21.957 | warn | Object sainlogic.0.weather.current.pressurerel is invalid: Default value has to be type "number" but received type "string"
sainlogic.0 | 2023-08-11 15:29:21.957 | warn | This object will not be created in future versions. Please report this to the developer.
sainlogic.0 | 2023-08-11 15:29:21.956 | warn | Object sainlogic.0.weather.current.indoorhumidity is invalid: Default value has to be type "number" but received type "string"
sainlogic.0 | 2023-08-11 15:29:21.949 | warn | This object will not be created in future versions. Please report this to the developer.
sainlogic.0 | 2023-08-11 15:29:21.948 | warn | Object sainlogic.0.weather.current.indoortemp is invalid: Default value has to be type "number" but received type "string"

</body>
</html>

phifogg, 2023-08-26:
Hm.. these errors should have been solved with 0.10.5. 
Let's keep this issue for leafwetness. Please open a new one for other observations if you still have them.

Petergoge, 2024-06-12:
Hi Phifogg, 
can you tell me when or if you include the leafwetness-sensor?
I still missing it a lot😢
Greetings
Daniel

Petergoge, 2024-06-12:
Thank you for adding the leafwetness.
I am not familiar with github and didn't saw that you already added it an August 26 2023 in the constants.js 🤭
I thought it would course a new version number, so I waited till today...Now I changed the file in my iobroker and there it is🥳

phifogg, 2024-06-27:
Updated in version 0.11.3

closed 2024-06-27 by phifogg

Petergoge, 2024-07-05:
Thank you for adding the sensor value in the newest version. 
Is it possible to add the battery information, too? That would be great!
Here is the debug information:

2024-07-05 15:41:28.961 - info: sainlogic.0 (14669) starting. Version 0.11.4 in /opt/iobroker/node_modules/iobroker.sainlogic, 
2024-07-05 15:41:29.010 - debug: sainlogic.0 (14669) Listener IP: 0.0.0.0
2024-07-05 15:41:29.011 - debug: sainlogic.0 (14669) Listener port: 49160
2024-07-05 15:41:29.011 - debug: sainlogic.0 (14669) Listener path: /wetterstation
2024-07-05 15:41:29.012 - debug: sainlogic.0 (14669) Listener protocol: ecowitt
2024-07-05 15:41:29.012 - debug: sainlogic.0 (14669) Forward URL:
2024-07-05 15:41:29.014 - info: sainlogic.0 (14669) Starting Listener
2024-07-05 15:41:50.280 - debug: sainlogic.0 (14669) Listener body is PASSKEY=41C6E12B83D8C7957969C590ECB89F63&stationtype=GW1100A_V2.3.3&runtime=5804&heap=25588&dateutc=2024-07-05+13:41:49&tempinf=79.70&humidityin=48&baromrelin=29.751&baromabsin=29.751&soilmoisture1=23&soilad1=159&leafwetness_ch1=0&soilbatt1=1.3&leaf_batt1=0.55&freq=868M&model=GW1100A&interval=60
2024-07-05 15:41:50.282 - debug: sainlogic.0 (14669) Listener received EW update: {"PASSKEY":"41C6E12B83D8C7957969C590ECB89F63","stationtype":"GW1100A_V2.3.3","runtime":"5804","heap":"25588","dateutc":"2024-07-05 13:41:49","tempinf":"79.70","humidityin":"48","baromrelin":"29.751","baromabsin":"29.751","soilmoisture1":"23","soilad1":"159","leafwetness_ch1":"0","soilbatt1":"1.3","leaf_batt1":"0.55","freq":"868M","model":"GW1100A","interval":"60"}
2024-07-05 15:41:50.404 - debug: sainlogic.0 (14669) No mapping found for dataelement: PASSKEY
2024-07-05 15:41:50.407 - debug: sainlogic.0 (14669) Extracting value for info.softwaretype(stationtype)
2024-07-05 15:41:50.408 - debug: sainlogic.0 (14669) No mapping found for dataelement: runtime
2024-07-05 15:41:50.408 - debug: sainlogic.0 (14669) No mapping found for dataelement: heap
2024-07-05 15:41:50.409 - debug: sainlogic.0 (14669) No mapping found for dataelement: dateutc
2024-07-05 15:41:50.409 - debug: sainlogic.0 (14669) Extracting value for weather.current.indoortemp(tempinf)
2024-07-05 15:41:50.417 - debug: sainlogic.0 (14669) Extracting value for weather.current.indoorhumidity(humidityin)
2024-07-05 15:41:50.417 - debug: sainlogic.0 (14669) Extracting value for weather.current.pressurerel(baromrelin)
2024-07-05 15:41:50.419 - debug: sainlogic.0 (14669) Extracting value for weather.current.pressureabs(baromabsin)
2024-07-05 15:41:50.420 - debug: sainlogic.0 (14669) Extracting value for weather.current.1.soilmoisture(soilmoisture1)
2024-07-05 15:41:50.420 - debug: sainlogic.0 (14669) No mapping found for dataelement: soilad1
2024-07-05 15:41:50.421 - debug: sainlogic.0 (14669) Extracting value for weather.current.leafwetness(leafwetness_ch1)
2024-07-05 15:41:50.421 - debug: sainlogic.0 (14669) Extracting value for weather.info.1.soilbatt(soilbatt1)
2024-07-05 15:41:50.422 - debug: sainlogic.0 (14669) No mapping found for dataelement: leaf_batt1
2024-07-05 15:41:50.422 - debug: sainlogic.0 (14669) Extracting value for info.freq(freq)
2024-07-05 15:41:50.423 - debug: sainlogic.0 (14669) Extracting value for info.model(model)
2024-07-05 15:41:50.424 - debug: sainlogic.0 (14669) No mapping found for dataelement: interval
2024-07-05 15:41:50.424 - debug: sainlogic.0 (14669) Extracting value for info.last_listener_update(last_listener_update)
2024-07-05 15:41:50.426 - debug: sainlogic.0 (14669) Setting value from data for info.softwaretype to GW1100A_V2.3.3
2024-07-05 15:41:50.429 - debug: sainlogic.0 (14669) Setting value from data for weather.current.indoortemp to 26.5
2024-07-05 15:41:50.430 - debug: sainlogic.0 (14669) Target for indoortemp unit is set: °C, using conversion: null
2024-07-05 15:41:50.431 - debug: sainlogic.0 (14669) Setting value from data for weather.current.indoorhumidity to 48
2024-07-05 15:41:50.431 - debug: sainlogic.0 (14669) Setting value from data for weather.current.pressurerel to 1007.4
2024-07-05 15:41:50.431 - debug: sainlogic.0 (14669) Target for pressurerel unit is set: hPa, using conversion: null
2024-07-05 15:41:50.432 - debug: sainlogic.0 (14669) Setting value from data for weather.current.pressureabs to 1007.4
2024-07-05 15:41:50.432 - debug: sainlogic.0 (14669) Target for pressureabs unit is set: hPa, using conversion: null
2024-07-05 15:41:50.433 - debug: sainlogic.0 (14669) Setting value from data for weather.current.1.soilmoisture to 23
2024-07-05 15:41:50.433 - debug: sainlogic.0 (14669) Setting value from data for weather.current.leafwetness to 0
2024-07-05 15:41:50.434 - debug: sainlogic.0 (14669) Setting value from data for weather.info.1.soilbatt to 1.3
2024-07-05 15:41:50.434 - debug: sainlogic.0 (14669) Setting value from data for info.freq to 868M
2024-07-05 15:41:50.434 - debug: sainlogic.0 (14669) Setting value from data for info.model to GW1100A
2024-07-05 15:41:50.435 - debug: sainlogic.0 (14669) Setting value from data for info.last_listener_update to {"PASSKEY":"41C6E12B83D8C7957969C590ECB89F63","stationtype":"GW1100A_V2.3.3","runtime":"5804","heap":"25588","dateutc":"2024-07-05 13:41:49","tempinf":"79.70","humidityin":"48","baromrelin":"29.751","baromabsin":"29.751","soilmoisture1":"23","soilad1":"159","leafwetness_ch1":"0","soilbatt1":"1.3","leaf_batt1":"0.55","freq":"868M","model":"GW1100A","interval":"60"}`