phifogg/ioBroker.sainlogic #181: Feature request: Weiterleitungs-URL auf andere IP:Port

Sundancer78, 2023-08-11:
Ist es möglich die Weiterleitungs-URL so einzurichten, dass die empfangenen Daten von dem Ecowitt Gateway auch an eine andere IP:Port weitergeleitet werden?

Ich habe neben dem ioBroker noch CumulusMX auf dem gleichen Rechner laufen und würde in CumulusMX auch weiterhin die Daten von Ecowitt empfangen.

phifogg, 2023-08-11:
Ja, das ist die Idee von dem Forwarder. Einfach die URL zu dem CumulusMX eintragen mit Port.

Sundancer78, 2023-08-12:
> Ja, das ist die Idee von dem Forwarder. Einfach die URL zu dem CumulusMX eintragen mit Port.

ähmmm, muss die URL ein bestimmtes Format haben? Ich hatte nur IP:Port eingetragen, ohne http etc. Danach kamen Fehlermeldungen im Protokoll vom Adapter. Ich dachte mir das es auf diese Weise nicht funktioniert und habe daraufhin die Anfrage erstellt.

2023-08-12 05:12:31.955 - error: sainlogic.0 (545124) uncaught exception: Invalid URL
2023-08-12 05:12:31.959 - error: sainlogic.0 (545124) TypeError [ERR_INVALID_URL]: Invalid URL
2023-08-12 05:12:31.960 - error: sainlogic.0 (545124) Exception-Code: ERR_INVALID_URL: Invalid URL
2023-08-12 05:12:31.964 - info: sainlogic.0 (545124) Sainlogic Adapter gracefully shut down...
2023-08-12 05:12:31.965 - info: sainlogic.0 (545124) terminating
2023-08-12 05:12:31.966 - warn: sainlogic.0 (545124) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-08-12 05:12:32.469 - info: sainlogic.0 (545124) terminating
2023-08-12 05:12:32.574 - error: host.iobroker instance system.adapter.sainlogic.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-08-12 05:12:32.574 - info: host.iobroker Restart adapter system.adapter.sainlogic.0 because enabled
2023-08-12 05:13:02.732 - info: host.iobroker instance system.adapter.sainlogic.0 started with pid 545153
2023-08-12 05:13:03.738 - info: sainlogic.0 (545153) starting. Version 0.10.5 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v18.17.0, js-controller: 5.0.11
2023-08-12 05:13:03.757 - info: sainlogic.0 (545153) Starting Listener
2023-08-12 05:13:13.966 - error: sainlogic.0 (545153) uncaught exception: Invalid URL
2023-08-12 05:13:13.970 - error: sainlogic.0 (545153) TypeError [ERR_INVALID_URL]: Invalid URL
2023-08-12 05:13:13.970 - error: sainlogic.0 (545153) Exception-Code: ERR_INVALID_URL: Invalid URL
2023-08-12 05:13:13.974 - info: sainlogic.0 (545153) Sainlogic Adapter gracefully shut down...
2023-08-12 05:13:13.975 - info: sainlogic.0 (545153) terminating
2023-08-12 05:13:13.976 - warn: sainlogic.0 (545153) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-08-12 05:13:14.478 - info: sainlogic.0 (545153) terminating
2023-08-12 05:13:14.580 - error: host.iobroker instance system.adapter.sainlogic.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-08-12 05:13:14.581 - info: host.iobroker Restart adapter system.adapter.sainlogic.0 because enabled
2023-08-12 05:13:44.760 - info: host.iobroker instance system.adapter.sainlogic.0 started with pid 545182
2023-08-12 05:13:45.850 - info: sainlogic.0 (545182) starting. Version 0.10.5 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v18.17.0, js-controller: 5.0.11
2023-08-12 05:13:45.869 - info: sainlogic.0 (545182) Starting Listener
2023-08-12 05:13:55.990 - error: sainlogic.0 (545182) uncaught exception: Invalid URL
2023-08-12 05:13:55.994 - error: sainlogic.0 (545182) TypeError [ERR_INVALID_URL]: Invalid URL
at new NodeError (node:internal/errors:405:5)
at new URL (node:internal/url:611:13)
at Server. (/opt/iobroker/node_modules/iobroker.sainlogic/lib/listener.js:114:49)
at Server.emit (node:events:514:28)
at parserOnIncoming (node:_http_server:1107:12)
at HTTPParser.parserOnHeadersComplete (node:_http_common:119:17)
2023-08-12 05:13:55.995 - error: sainlogic.0 (545182) Exception-Code: ERR_INVALID_URL: Invalid URL
2023-08-12 05:13:56.000 - info: sainlogic.0 (545182) Sainlogic Adapter gracefully shut down...
2023-08-12 05:13:56.001 - info: sainlogic.0 (545182) terminating
2023-08-12 05:13:56.001 - warn: sainlogic.0 (545182) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-08-12 05:13:56.503 - info: sainlogic.0 (545182) terminating
2023-08-12 05:13:56.592 - error: host.iobroker instance system.adapter.sainlogic.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-08-12 05:13:56.592 - info: host.iobroker Restart adapter system.adapter.sainlogic.0 because enabled
2023-08-12 05:13:56.592 - warn: host.iobroker Do not restart adapter system.adapter.sainlogic.0 because restart loop detected

phifogg, 2023-08-13:
ja, die muss mit http://<hostname>:<port>/<Ggf. pfad> eingetragen werden.

phifogg, 2023-08-26:
ich schliesse das hier mal. wenn es doch noch probleme gibt bitte wieder öffnen.

closed 2023-08-26 by phifogg