phifogg/ioBroker.sainlogic #184: Listener received illegal request

easybeat, 2023-11-14:
**Describe the bug**  
I have a WS3500 and receive a message

Listener received illegal request: /weatherstation/updateweatherstation.phpID=1234&PASSWORD=ertg&indoortempf=74.8&tempf=48.9&dewptf=46.9&windchillf=48.9&indoorhumidity=40&humidity=93&windspeedmph=0.7&windgustmph=1.1&winddir=83&absbaromin=27.673&baromin=30.042&rainin=0.118&dailyrainin=0.331&weeklyrainin=0.890&monthlyrainin=2.890&solarradiation=0.34&UV=0&dateutc=2023-11-14%2006:38:56&softwaretype=EasyWeatherV1.6.6&action=updateraw&realtime=1&rtfreq=5

**To Reproduce**  
Steps to reproduce the behavior:
1. Setup my weatherstation to Wunderground (see screenshot)
2. Setup the Adapter to same settings (see screenshot)
3. Startet the instance
4. Check Protocoll on ioBroker

**Expected behavior**  
A clear and concise description of what you expected to happen.

**Screenshots & Logfiles**  
[image]
[image]

**Versions:**  
 - Adapter version: 0.10.5
 - JS-Controller version: 5.0.12
 - Node version: 18.18.2
 - Operating system: Ubuntu 22.04 Server
**Additional context**  
Add any other context about the problem here.

phifogg, 2023-11-14:
Hi Easybeat,
try adding a '?' in the setup of the weatherstation in the path option. Only on the station, not in the adapter.
Daniel

easybeat, 2023-11-14:
Hi Daniel
thank you very much. This works now.
Kind regards
Beat

closed 2023-11-14 by easybeat