phifogg/ioBroker.sainlogic #186: Missing objects feelslike, hourlyrain, eventrain...

nicmacsoem, 2023-12-22:
Hallo,

ich migriere aktuell die Integration meiner Wetterstation (WH2650A+Innensensor+Bodenfeuchtesensor) von einem bisherigen UDP-Plugin (LoxBerry) auf diesen ioBroker-Adapter.
Dabei ist mir aufgefallen, dass im Vgl. zur WS-View-App folgende Objekte/Wetterdaten der Station fehlen:
- feelslike
- hourlyrain
- eventrain
(- wh65batt = Batteriestatus des Innensensors lt. vorher genutztem Plugin)
Könnten diese bitte noch implementiert werden?!

Danke & Grüße

Nico

phifogg, 2024-02-02:
Hallo Nico,
kannst Du mal ein Log schicken wenn die Station Daten schickt? Dann schaue ich mir das an, sollte nicht so schwer sein.

nicmacsoem, 2024-02-04:
Hallo Daniel,

nachfolgend die Log-Ausgabe, bei der bis auf Feelslike offenbar auch die fehlenden Parameter von der Station gesendet werden:
`2024-02-04 15:04:28.808 - debug: sainlogic.0 (104575) Listener body is PASSKEY=xxxxxxxxxxxxx&stationtype=WH2650A_V1.7.7&runtime=9166419&dateutc=2024-02-04+14:04:28&tempinf=46.58&humidityin=89&baromrelin=32.123&baromabsin=29.560&tempf=50.00&humidity=88&winddir=168&windspeedmph=7.16&windgustmph=13.65&maxdailygust=27.29&solarradiation=36.82&uv=0&rainratein=0.000&eventrainin=0.260&hourlyrainin=0.008&dailyrainin=0.130&weeklyrainin=0.130&monthlyrainin=0.339&yearlyrainin=0.339&totalrainin=0.339&temp1f=2210.72&temp2f=1512.32&temp3f=4641.80&temp4f=3355.52&wh65batt=0&wh25batt=0&freq=868M&model=WH2650A
2024-02-04 15:04:28.814 - debug: sainlogic.0 (104575) Listener received EW update: {"PASSKEY":"xxxxxxxxxxxxx","stationtype":"WH2650A_V1.7.7","runtime":"9166419","dateutc":"2024-02-04 14:04:28","tempinf":"46.58","humidityin":"89","baromrelin":"32.123","baromabsin":"29.560","tempf":"50.00","humidity":"88","winddir":"168","windspeedmph":"7.16","windgustmph":"13.65","maxdailygust":"27.29","solarradiation":"36.82","uv":"0","rainratein":"0.000","eventrainin":"0.260","hourlyrainin":"0.008","dailyrainin":"0.130","weeklyrainin":"0.130","monthlyrainin":"0.339","yearlyrainin":"0.339","totalrainin":"0.339","temp1f":"2210.72","temp2f":"1512.32","temp3f":"4641.80","temp4f":"3355.52","wh65batt":"0","wh25batt":"0","freq":"868M","model":"WH2650A"}
2024-02-04 15:04:28.860 - debug: sainlogic.0 (104575) No mapping found for dataelement: PASSKEY
2024-02-04 15:04:28.869 - debug: sainlogic.0 (104575) Extracting value for info.softwaretype(stationtype)
2024-02-04 15:04:28.871 - debug: sainlogic.0 (104575) No mapping found for dataelement: runtime
2024-02-04 15:04:28.873 - debug: sainlogic.0 (104575) No mapping found for dataelement: dateutc
2024-02-04 15:04:28.874 - debug: sainlogic.0 (104575) Extracting value for weather.current.indoortemp(tempinf)
2024-02-04 15:04:28.888 - debug: sainlogic.0 (104575) Extracting value for weather.current.indoorhumidity(humidityin)
2024-02-04 15:04:28.890 - debug: sainlogic.0 (104575) Extracting value for weather.current.pressurerel(baromrelin)
2024-02-04 15:04:28.893 - debug: sainlogic.0 (104575) Extracting value for weather.current.pressureabs(baromabsin)
2024-02-04 15:04:28.897 - debug: sainlogic.0 (104575) Extracting value for weather.current.outdoortemp(tempf)
2024-02-04 15:04:28.900 - debug: sainlogic.0 (104575) Extracting value for weather.current.outdoorhumidity(humidity)
2024-02-04 15:04:28.902 - debug: sainlogic.0 (104575) Extracting value for weather.current.winddir(winddir)
2024-02-04 15:04:28.903 - debug: sainlogic.0 (104575) Extracting value for weather.current.windspeed(windspeedmph)
2024-02-04 15:04:28.907 - debug: sainlogic.0 (104575) Extracting value for weather.current.windgustspeed(windgustmph)
2024-02-04 15:04:28.910 - debug: sainlogic.0 (104575) Extracting value for weather.current.maxdailygust(maxdailygust)
2024-02-04 15:04:28.913 - debug: sainlogic.0 (104575) Extracting value for weather.current.solarradiation(solarradiation)
2024-02-04 15:04:28.915 - debug: sainlogic.0 (104575) Extracting value for weather.current.uvi(uv)
2024-02-04 15:04:28.916 - debug: sainlogic.0 (104575) Extracting value for weather.current.rain(rainratein)
2024-02-04 15:04:28.919 - debug: sainlogic.0 (104575) No mapping found for dataelement: eventrainin
2024-02-04 15:04:28.920 - debug: sainlogic.0 (104575) No mapping found for dataelement: hourlyrainin
2024-02-04 15:04:28.922 - debug: sainlogic.0 (104575) Extracting value for weather.current.dailyrain(dailyrainin)
2024-02-04 15:04:28.925 - debug: sainlogic.0 (104575) Extracting value for weather.current.weeklyrain(weeklyrainin)
2024-02-04 15:04:28.928 - debug: sainlogic.0 (104575) Extracting value for weather.current.monthlyrain(monthlyrainin)
2024-02-04 15:04:28.930 - debug: sainlogic.0 (104575) Extracting value for weather.current.yearlyrain(yearlyrainin)
2024-02-04 15:04:28.932 - debug: sainlogic.0 (104575) Extracting value for weather.current.totalrainin(totalrainin)
2024-02-04 15:04:28.934 - debug: sainlogic.0 (104575) Extracting value for weather.current.1.outdoortemp(temp1f)
2024-02-04 15:04:28.936 - debug: sainlogic.0 (104575) Extracting value for weather.current.2.outdoortemp(temp2f)
2024-02-04 15:04:28.937 - debug: sainlogic.0 (104575) Extracting value for weather.current.3.outdoortemp(temp3f)
2024-02-04 15:04:28.939 - debug: sainlogic.0 (104575) Extracting value for weather.current.4.outdoortemp(temp4f)
2024-02-04 15:04:28.941 - debug: sainlogic.0 (104575) No mapping found for dataelement: wh65batt
2024-02-04 15:04:28.942 - debug: sainlogic.0 (104575) Extracting value for weather.info.wh25batt(wh25batt)
2024-02-04 15:04:28.943 - debug: sainlogic.0 (104575) Extracting value for info.freq(freq)
2024-02-04 15:04:28.944 - debug: sainlogic.0 (104575) Extracting value for info.model(model)
2024-02-04 15:04:28.945 - debug: sainlogic.0 (104575) Extracting value for info.last_listener_update(last_listener_update)
2024-02-04 15:04:28.947 - debug: sainlogic.0 (104575) Setting value from data for info.softwaretype to WH2650A_V1.7.7
2024-02-04 15:04:28.948 - debug: sainlogic.0 (104575) Setting value from data for weather.current.indoortemp to 8.1
2024-02-04 15:04:28.950 - debug: sainlogic.0 (104575) Target for indoortemp unit is set: °C, using conversion: null
[... cut ...] :"32.123","baromabsin":"29.560","tempf":"50.00","humidity":"88","winddir":"168","windspeedmph":"7.16","windgustmph":"13.65","maxdailygust":"27.29","solarradiation":"36.82","uv":"0","rainratein":"0.000","eventrainin":"0.260","hourlyrainin":"0.008","dailyrainin":"0.130","weeklyrainin":"0.130","monthlyrainin":"0.339","yearlyrainin":"0.339","totalrainin":"0.339","temp1f":"2210.72","temp2f":"1512.32","temp3f":"4641.80","temp4f":"3355.52","wh65batt":"0","wh25batt":"0","freq":"868M","model":"WH2650A"}`

phifogg, 2024-06-27:
Updated in version 0.11.3

closed 2024-06-27 by phifogg

nicmacsoem, 2024-06-30:
Hallo Daniel & danke dir für den Fix. Leider lässt sich der iOBroker-Adapter nicht per admin auf die 0.11.3 updaten - als aktuellste Version wird mir weiterhin 0.10.5 angezeigt. Aktualisiere ich "manuell" via NPM erhalte ich die aktuellste Version & im Log mehrere Warnungen die drei neuen Werte betreffend:
[image]
Soll das so sein? Der wh65batt ist nach meiner bisherigen Erfahrung nur ein "Schaltausgang" - nach internem/unbekanntem Schwellwert wechselt der bei kritischem Batteriezustand von 0 auf 1 (also keine Spannungswert wie bei den anderen).
Kannst du da nochmal schauen?
Danke & viele Grüße
Nico