phifogg/ioBroker.sainlogic #190: When restarting, all data points are set to 0 / Bei Neustart werden alle Datenpunkte auf 0 gesetzt

sansonifabio, 2024-02-03:
**Describe the bug**  

Hello, if you restart the adapter or the server, the adapter sets all 
Data points set to 0. For me, for example, this triggers certain scripts that trigger at 0 degrees. 
Can't you do it so that the adapter leaves the data points as they were before the restart and then updates them in the next interval?

Hallo @phifogg wenn man den Adapter neu startet, oder auch den Server, setzt der Adapter alle
Datenpunkte auf 0. 
Das löst dann bei mir zum Beispiel, bestimmte Scripte aus die bei 0 Grad triggern. 

Kann man es nicht so machen, das der Adapter die Datenpunkte so beläßt wie sie vor dem Neustart waren und dann im nächsten Intervall dann aktualisiert?

Herzliche Grüße
Fabio

**To Reproduce**  

1. Restart Adapter or Server

**Versions:**  
 - Adapter version: 0.10.5
 - JS-Controller version: 5.0.17
 - Node version: 18.19.0
 - Operating system: Debian Bookworm
 - NPM: 10.2.3

phifogg, 2024-06-27:
Ich schau's mir an.

phifogg, 2024-11-12:
Kannst Du mir mal Debug-Logs von Dir schicken? Ich habe das bei mir ausprobiert, hier werden die Werte nicht gelöscht.

sansonifabio, 2024-11-13:
@phifogg sehr interessant hab ja lange nicht meher kontrolliert heutet hat er die datenpunkte gelassen.
Herzliche Grüße
Fabio

`<html>
<body>

sainlogic.0 | 2024-11-13 12:44:57.473 | debug | FW Scheduler Connection closed
-- | -- | -- | --
sainlogic.0 | 2024-11-13 12:44:57.469 | debug | Target for totalrainin unit is set: mm, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.469 | debug | Setting value from data for weather.current.totalrainin to 1921.1
sainlogic.0 | 2024-11-13 12:44:57.469 | debug | Target for yearlyrain unit is set: mm, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.469 | debug | Setting value from data for weather.current.yearlyrain to 913.4
sainlogic.0 | 2024-11-13 12:44:57.469 | debug | Target for windspeed unit is set: km/h, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.469 | debug | Setting value from data for weather.current.windspeed to 0.4
sainlogic.0 | 2024-11-13 12:44:57.469 | debug | Target for windgustspeed unit is set: km/h, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.468 | debug | Setting value from data for weather.current.windgustspeed to 1.8
sainlogic.0 | 2024-11-13 12:44:57.468 | debug | GetHeading returning: WNW
sainlogic.0 | 2024-11-13 12:44:57.468 | debug | Determining wind heading for 290 and precision 16
sainlogic.0 | 2024-11-13 12:44:57.468 | debug | Setting value from data for weather.current.winddir to 290
sainlogic.0 | 2024-11-13 12:44:57.468 | debug | Target for windchilltemp unit is set: °C, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.467 | debug | Setting value from data for weather.current.windchilltemp to 8.3
sainlogic.0 | 2024-11-13 12:44:57.467 | debug | Target for weeklyrain unit is set: mm, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.467 | debug | Setting value from data for weather.current.weeklyrain to 8.4
sainlogic.0 | 2024-11-13 12:44:57.467 | debug | Setting value from data for weather.current.uvi to 0
sainlogic.0 | 2024-11-13 12:44:57.467 | debug | Target for solarradiation unit is set: lx, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.467 | debug | Setting value from data for weather.current.solarradiation to 8680
sainlogic.0 | 2024-11-13 12:44:57.466 | debug | Target for rain unit is set: mm, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.465 | debug | Setting value from data for weather.current.rain to 0
sainlogic.0 | 2024-11-13 12:44:57.465 | debug | Target for pressurerel unit is set: hPa, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.465 | debug | Setting value from data for weather.current.pressurerel to 1026.2
sainlogic.0 | 2024-11-13 12:44:57.465 | debug | Target for pressureabs unit is set: hPa, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.465 | debug | Setting value from data for weather.current.pressureabs to 1026.2
sainlogic.0 | 2024-11-13 12:44:57.464 | debug | Target for outdoortemp unit is set: °C, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.464 | debug | Setting value from data for weather.current.outdoortemp to 8.3
sainlogic.0 | 2024-11-13 12:44:57.464 | debug | Setting value from data for weather.current.outdoorhumidity to 89
sainlogic.0 | 2024-11-13 12:44:57.464 | debug | Target for monthlyrain unit is set: mm, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.464 | debug | Setting value from data for weather.current.monthlyrain to 9.1
sainlogic.0 | 2024-11-13 12:44:57.463 | debug | Target for indoortemp unit is set: °C, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.463 | debug | Setting value from data for weather.current.indoortemp to 22.7
sainlogic.0 | 2024-11-13 12:44:57.463 | debug | Setting value from data for weather.current.indoorhumidity to 43
sainlogic.0 | 2024-11-13 12:44:57.463 | debug | Target for dewpointtemp unit is set: °C, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.462 | debug | Setting value from data for weather.current.dewpointtemp to 6.6
sainlogic.0 | 2024-11-13 12:44:57.462 | debug | Target for dailyrain unit is set: mm, using conversion: null
sainlogic.0 | 2024-11-13 12:44:57.462 | debug | Setting value from data for weather.current.dailyrain to 0
sainlogic.0 | 2024-11-13 12:44:57.461 | debug | Setting value from data for weather.current.UVraw to 67
sainlogic.0 | 2024-11-13 12:44:57.449 | info | Scheduler updating IOBroker states
sainlogic.0 | 2024-11-13 12:44:57.448 | debug | Data object: {"indoortemp":227,"outdoortemp":83,"dewpointtemp":66,"windchilltemp":83,"heatindex":83,"indoorhumidity":43,"outdoorhumidity":89,"pressureabs":10262,"pressurerel":10262,"winddir":290,"windspeed":1,"windgustspeed":5,"rain":0,"dailyrain":0,"weeklyrain":84,"monthlyrain":91,"yearlyrain":9134,"totalrainin":19211,"solarradiation":86800,"UVraw":67,"uvi":0}
sainlogic.0 | 2024-11-13 12:44:57.446 | debug | Data Command received: 11 subcommand 4
sainlogic.0 | 2024-11-13 12:44:57.441 | debug | FW Scheduler Received data string: ffff0b0050040100e3020053030042040053050053062b07590828160928160a01220b00010c00050e0000000010000000001100000054120000005b13000023ae1400004b0b15000153101600431700c0db
sainlogic.0 | 2024-11-13 12:44:57.428 | debug | Scheduler connected to weather station run 255,255,11,0,6,4,4,25
sainlogic.0 | 2024-11-13 12:44:57.419 | info | Scheduler pulling for new data
sainlogic.0 | 2024-11-13 12:43:57.416 | debug | Scheduler Interval: 60
sainlogic.0 | 2024-11-13 12:43:57.415 | debug | Weatherstation port: 45000
[... cut ...]
admin.0 | 2024-11-13 12:43:56.412 | info | <== Disconnect system.user.admin from ::ffff:192.168.2.43
host.nuc | 2024-11-13 12:43:53.583 | info | instance system.adapter.sainlogic.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
host.nuc | 2024-11-13 12:43:53.011 | info | stopInstance system.adapter.sainlogic.0 send kill signal
host.nuc | 2024-11-13 12:43:52.965 | info | stopInstance system.adapter.sainlogic.0 (force=false, process=true)
admin.0 | 2024-11-13 12:43:43.297 | info | ==> Connected system.user.admin from ::ffff:192.168.2.43

</body>
</html>`

phifogg, 2024-11-13:
Interessant. Aber gut zu hören, dass das Problem nicht mehr existiert. 

Es kann durchaus sein, dass ich das früher mal drin hatte. Da hat der Adapter bei jedem Neustart alle Datenpunkte gepüft. Das mache ich nun nicht mehr. Ich schliesse das Issue mal, sollte es wieder auftreten melde Dich.

closed 2024-11-13 by phifogg