phifogg/ioBroker.sainlogic #191: Scheduler received data for unkown command

GameWatchProjects, 2024-02-12:
**Describe the bug**  
sainlogic.0
	2024-02-12 13:35:11.213	error	Scheduler received data for unkown command 91 subcommand 97
sainlogic.0
	2024-02-12 13:35:11.202	error	Scheduler received data for unkown command 80 subcommand 71

Ecowitt Wlan-USB Adapter GW1101 (Steht so tatsächlich auf dem Gehäuse) mit Firmware Version GW1100A_V2.3.1

Listener funktioniert einwandfrei, Scheduler wirft oben genannten Fehler aus ...

phifogg, 2024-06-27:
Der Scheduler funktioniert nur mit bestimmten Wetterstationen. Für andere habe ich leider keine Doku wie genau die anzusprechen wären.

GameWatchProjects, 2024-06-28:
Hallo, kein Problem, hatte wegen der ausbleibenden Resonanz zwischenzeitlich den Adapter entfernt und die vorhandene Wetterstation über weewx und MQTT erfolgreich in IoBroker eingebunden, so dass die Meldungen der Wetterstation ebenfalls in Echtzeit nun erscheinen. Dieser Hinweis wäre sicherlich auch für andere Wettertstaionsbetreiber hilfreich, welche ihre Station ebenfalls nicht sauber mit diesem Adapter in IoBroker eingebunden bekommen.

closed 2024-06-28 by GameWatchProjects