phifogg/ioBroker.sainlogic #192: CO2 sensor data not mapped propperlz

juergsiw, 2024-02-12:
**Describe the bug**  
The data fields provided by a CO2 sensor are not mapped propperly. 
e.g. co2_batt is mapped to co2 
So the co2 state always has value 0 although different values are provided by the Sensor/Wheather-Station. Details see logs below.

**To Reproduce**  
nothing specific, simply run latest adapter version 

**Expected behavior**  
Each provided data field shall be mapped exclusively to it's exact corresponding object.
pm10_24h_co2 -> pm10_24h_co2
co2 -> co2
co2_24h -> co2_24h
co2_batt -> co2_batt

**Root Cause**  
Inprecise reqexp in constant.js:
```
...
 id: 'pm10_24h_co2',
...
 wunderground: '^pm10_co2',  --> '^pm10_24h_co2'
 ecowitt: '^pm10_co2',       --> '^pm10_24h_co2'
...
 id: 'co2',
...
 wunderground: 'co2', --> '^co2$'
 ecowitt: 'co2', --> '^co2$'
```

**Screenshots & Logfiles**  
...
_**2024-02-12 20:06:31.167	debug	Extracting value for weather.current.co2(co2_batt)**_
...
_**2024-02-12 20:06:31.165	debug	Extracting value for weather.current.co2(co2_24h)**_
_**2024-02-12 20:06:31.165	debug	Extracting value for weather.current.co2(co2)**_
2024-02-12 20:06:31.165	debug	Extracting value for weather.daily.pm25_24h_co2(pm25_24h_co2)
2024-02-12 20:06:31.165	debug	Extracting value for weather.current.pm25_co2(pm25_co2)
_**2024-02-12 20:06:31.165	debug	Extracting value for weather.current.co2(pm10_24h_co2)**_
2024-02-12 20:06:31.165	debug	Extracting value for weather.current.pm10_co2(pm10_co2)
2024-02-12 20:06:31.164	debug	Extracting value for weather.current.humi_co2(humi_co2)
2024-02-12 20:06:31.164	debug	Extracting value for weather.current.tf_co2(tf_co2)
...
2024-02-12 20:06:31.153	debug	Listener received EW update: {"PASSKEY":"BDE1491A73DF1A2A3779070D90D8F929","stationtype":"EasyWeatherV1.5.9","dateutc":"2024-02-12 19:06:28","tempinf":"72.1","humidityin":"44","baromrelin":"29.811","baromabsin":"28.412","tempf":"43.3","humidity":"90","winddir":"92","winddir_avg10m":"123","windspeedmph":"1.3","windspdmph_avg10m":"0.9","windgustmph":"2.2","maxdailygust":"20.6","rainratein":"0.000","eventrainin":"0.000","hourlyrainin":"0.000","dailyrainin":"0.020","weeklyrainin":"0.071","monthlyrainin":"0.480","yearlyrainin":"2.543","solarradiation":"0.00","uv":"0","temp1f":"70.2","humidity1":"46","temp2f":"70.2","humidity2":"47","temp3f":"70.0","humidity3":"53","temp4f":"46.0","humidity4":"79","temp5f":"64.9","humidity5":"50","temp6f":"58.8","humidity6":"63","temp7f":"71.2","humidity7":"44","temp8f":"70.9","humidity8":"46","tf_co2":"72.7","humi_co2":"52","pm10_co2":"1.9","pm10_24h_co2":"2.0","pm25_co2":"1.9","pm25_24h_co2":"1.6","co2":"695","co2_24h":"675","wh65batt":"0","wh25batt":"0","batt1":"0","batt2":"0","batt3":"0","batt4":"0","batt5":"0","batt6":"0","batt7":"0","batt8":"0","co2_batt":"0","freq":"868M","model":"HP2550A_Pro_V1.7.2"}
2024-02-12 20:06:31.153	debug	Listener body is PASSKEY=BDE1491A73DF1A2A3779070D90D8F929&stationtype=EasyWeatherV1.5.9&dateutc=2024-02-12+19:06:28&tempinf=72.1&humidityin=44&baromrelin=29.811&baromabsin=28.412&tempf=43.3&humidity=90&winddir=92&winddir_avg10m=123&windspeedmph=1.3&windspdmph_avg10m=0.9&windgustmph=2.2&maxdailygust=20.6&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.020&weeklyrainin=0.071&monthlyrainin=0.480&yearlyrainin=2.543&solarradiation=0.00&uv=0&temp1f=70.2&humidity1=46&temp2f=70.2&humidity2=47&temp3f=70.0&humidity3=53&temp4f=46.0&humidity4=79&temp5f=64.9&humidity5=50&temp6f=58.8&humidity6=63&temp7f=71.2&humidity7=44&temp8f=70.9&humidity8=46&tf_co2=72.7&humi_co2=52&pm10_co2=1.9&pm10_24h_co2=2.0&pm25_co2=1.9&pm25_24h_co2=1.6&co2=695&co2_24h=675&wh65batt=0&wh25batt=0&batt1=0&batt2=0&batt3=0&batt4=0&batt5=0&batt6=0&batt7=0&batt8=0&co2_batt=0&freq=868M&model=HP2550A_Pro_V1.7.2

**Versions:**  
 - Adapter version: 0.10.5
 - JS-Controller version: 5.0.17
 - Node version: v18.19.0
 - Operating system: 
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

phifogg, 2024-07-03:
Published 0.11.4 with a fix on this to NPM

phifogg, 2024-11-12:
closing as no more comments.

closed 2024-11-12 by phifogg