phifogg/ioBroker.sainlogic #195: Anzeige der Daten von DP70-Feuchtesensoren

frickelmeister, 2024-05-25:
Hallo,

ich betreibe zwei Feuchtesensoren DP70 an meiner froggit-Wetterstation. Diese werden aber im ioBroker nicht angezeigt.

Der Datenstring, den die Wetterstation schickt:
2024-05-25 19:51:06.854 - debug: sainlogic.0 (102074) Listener body is PASSKEY=XXX&stationtype=GW1100A_V2.3.2&runtime=260070&heap=24852&dateutc=2024-05-25+17:51:06&tempinf=69.44&humidityin=58&baromrelin=28.405&baromabsin=28.405&tempf=64.22&humidity=73&winddir=303&windspeedmph=0.00&windgustmph=0.00&maxdailygust=4.47&solarradiation=39.01&uv=0&rainratein=0.000&eventrainin=0.000&hourlyrainin=0.000&dailyrainin=0.012&weeklyrainin=4.122&monthlyrainin=6.319&yearlyrainin=16.260&totalrainin=16.260&temp1f=79.34&humidity1=40&temp2f=70.34&humidity2=58&temp3f=68.90&humidity3=60&soilmoisture1=48&soilad1=244&soilmoisture2=32&soilad2=192&lightning_num=0&lightning=&lightning_time=&leak_ch1=0&leak_ch3=0&tf_ch1=57.56&wh65batt=0&batt1=0&batt2=0&batt3=0&soilbatt1=1.5&soilbatt2=1.6&wh57batt=5&leakbatt1=3&leakbatt3=3&tf_batt1=1.46&freq=868M&model=GW1100A&interval=60

Interessant sind
- leak_ch1, leak_ch3 (=0 wenn kein Leck erkannt) - es gibt wohl auch noch leak_ch2 und leak_ch4, aber ich habe nur zwei Sensoren auf CH1 und CH3,
- leakbatt1, leakbatt3 (Batteriestand, ein Wert zwischen 0 und 5 und entspricht der Anzahl der "Striche", die in der WS View-App angezeigt werden) - leakbatt2 und leakbatt4 müsste es wenn die Sensoren vorhanden sind auch geben,
- wh57batt (das ist der Batteriestand des Blitzsensors, der ja im ioBroker angezeigt wird, aber nicht der Batteriestand).

Vielen Dank!

phifogg, 2024-06-27:
Updated in version 0.11.3

closed 2024-06-27 by phifogg

frickelmeister, 2024-06-27:
Thanks a lot!