phifogg/ioBroker.sainlogic #203: Update stable version in repo from 0.10.5 to 0.11.4

GermanBluefox, 2024-07-19:
# Think about update stable version to 0.11.4
**Version**: stable=**0.10.5** (361 days old) => latest=**0.11.4** (15 days old)
**Installs**: stable=**936** (79.52%), latest=**79** (6.71%), total=**1177**

Click to use [developer portal](https://www.iobroker.dev/adapter/sainlogic/releases)
Click to [edit](https://github.com/ioBroker/ioBroker.repositories/edit/master/sources-dist-stable.json#L2207)

**Do not close this issue manually as a new issue will be created if condition for update still exists.**

Please drop a comment if any reason exists which blocks updating to version 0.11.4 at this time.

Note: This is an automatically generated message and not personally authored by bluefox!
      @mcm1957 for evidence

mcm1957, 2024-07-28:
Is there any reason to delay updating stable any longer?
Please drop a commit if you know any reason to refrain from updating.

**If there is no response until11.8.2024 we could decide to update anyway.**

mcm1957, 2024-07-28:
README and news must be fixed
see https://github.com/phifogg/ioBroker.sainlogic/issues/204

closed 2024-07-30 by phifogg