phifogg/ioBroker.sainlogic #216: Forwarding still not working?

DieWalge, 2024-11-04:
**Describe the bug**  
Forwarding Weather Data ist still not working for me: sender logs look good but the receiving side only logs an empty update
`sainlogic.2 2024-11-04 10:53:05.589 debug Listener received EW update: {} ` (logs: see below).

**To Reproduce**  
ioBroker A: Set Forward-URL to Server B '
http://192.168.B.B:45002/weatherstation/updateweatherstation.php

ioBroker B: Listen-Port =45002, same URL und Protocol=EcoWitt

**Expected behavior**  
Server A should receive the Weather Data and forward everything to Server B, Server B should receive data and update ioBroker objects accordingly

**Actual behavior**
Server A:  all Weather Data ioBroker-Objects are up to date and the logfile indicates forwarding
Server B: forwarding is received but with no data. ioBroker objects are not updated

**Screenshots & Logfiles**  
`
**Server A (Forwarder):**

sainlogic.1 2024-11-04 10:52:04.730	debug	Forward response body: ok
sainlogic.1  2024-11-04 10:52:04.729	debug	Forward URL: + http://192.168.B.B:45002/weatherstation/updateweatherstation.php
sainlogic.1 2024-11-04 10:52:04.561	debug	Setting value from data for info.last_listener_update to {"PASSKEY":"***","stationtype":"EasyWeatherPro_V5.1.6","runtime":"1790165","heap":"24176","dateutc":"2024-11-04 09:52:03","tempinf":"73.8","humidityin":"61","baromrelin":"29.743","baromabsin":"29.743","tempf":"70.5","humidity":"78","winddir":"127","windspeedmph":"5.59","windgustmph":"8.05","maxdailygust":"8.05","solarradiation":"339.20","uv":"3","rainratein":"0.000","eventrainin":"0.000","hourlyrainin":"0.000","dailyrainin":"0.000","weeklyrainin":"0.000","monthlyrainin":"0.000","yearlyrainin":"0.512","temp1f":"72.1","humidity1":"82","soilmoisture1":"27","wh65batt":"0","batt1":"0","soilbatt1":"1.5","freq":"868M","model":"HP2561AE_Pro_V1.9.3","interval":"60"}
sainlogic.1 2024-11-04 10:52:04.561	debug	Setting value from data for info.model to HP2561AE_Pro_V1.9.3
...
sainlogic.1 2024-11-04 10:47:51.423	info	Starting Listener
sainlogic.1 2024-11-04 10:47:51.421	debug	Forward URL: http://192.168.B.B:45002/weatherstation/updateweatherstation.php
sainlogic.1 2024-11-04 10:47:51.421	debug	Listener protocol: ecowitt
sainlogic.1 2024-11-04 10:47:51.420	debug	Listener path: /weatherstation/updateweatherstation.php
sainlogic.1 2024-11-04 10:47:51.420	debug	Listener port: 45001
sainlogic.1 2024-11-04 10:47:51.418	debug	Listener IP: 0.0.0.0
sainlogic.1 2024-11-04 10:47:51.396	info	starting. Version 0.11.4 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v18.20.4, js-controller: 6.0.11`

**Server B (Receiver):**

sainlogic.2 2024-11-04 10:53:05.592	debug	Setting value from data for info.last_listener_update to {}
sainlogic.2 2024-11-04 10:53:05.592	debug	Extracting value for info.last_listener_update(last_listener_update)
**sainlogic.2 2024-11-04 10:53:05.589	debug	Listener received EW update: {}**
sainlogic.2 2024-11-04 10:53:05.589	debug	Listener body is
...
sainlogic.2 2024-11-04 10:47:35.285	info	Starting Listener
sainlogic.2 2024-11-04 10:47:35.285	debug	Forward URL:
sainlogic.2 2024-11-04 10:47:35.285	debug	Listener protocol: ecowitt
sainlogic.2 2024-11-04 10:47:35.285	debug	Listener path: /weatherstation/updateweatherstation.php
sainlogic.2 2024-11-04 10:47:35.285	debug	Listener port: 45002
sainlogic.2 2024-11-04 10:47:35.284	debug	Listener IP: 192.168.B.B
sainlogic.2 2024-11-04 10:47:35.275	info	starting. Version 0.11.4 in /opt/iobroker/node_modules/iobroker.sainlogic, node: v20.18.0, js-controller: 6.0.11

phifogg, 2024-11-12:
Thanks for the logs. I believe I found the issue. Forwarding was happening to early when the body wasn't yet ready.
Anyway, install from dev-branch to check if you want. Version should be 0.11.5

closed 2024-11-13 by DieWalge

phifogg, 2024-11-13:
Awesome, thanks for confirmation.

I won't promise, but please post your findings :D

DieWalge, 2024-11-13:
thanks! see: https://github.com/phifogg/ioBroker.sainlogic/issues/218