phifogg/ioBroker.sainlogic #218: Support for EcoWitt WFC01 (Smart Water Timer)

DieWalge, 2024-11-13:
### I would very much like to see support for the EcoWitt WFC01 (Smart Water Timer) in the Sainlogic Adapter.

**In short:** the EcoWitt WFC01 is an electric valve that is RF-controlled as an IoT Device via a Gateway like the Ecowitt GW2000. It can be operated via the EcoWitt App (Manual & Timer) and reports Data like "Current Water Flow" and "Water Temperature". A Gateway can control several WFC01 Units so you can build your own irrigation system. If you want to integrate a WFC01 into ioBroker that should be possible right now via an API (not tested yet) with custom URLs and parsing return values.

What I would like to see is a native integration of the WCF01s into the ioBroker device tree with special datapoints so that you can control the valve state and read the additional values like water-flow and water-temperature.

### Further reading:
Thanks to [olicat] and [Gyvate] from https://www.wetterstationsforum.info/

**Product-Overview:**
German: 
https://www.wetterstationsforum.info/wiki/doku.php?id=wiki:wetterstationen:ecowitt-stationen#ecowitt_wfc01_-_intelligentes_iot_wasserventil_-_wittflow
https://www.wetterstationsforum.info/viewtopic.php?t=1788
English: https://www.wxforum.net/index.php?PHPSESSID=rmmh8mt2nfi5lrbco5vqueutdg&topic=45657.msg462951#msg462951

**API:** https://www.wetterstationsforum.info/wiki/lib/exe/fetch.php?media=wiki:wetterstationen:local_iot_api_20240528.pdf
**Architecture:** https://www.wetterstationsforum.info/wiki/doku.php?id=wiki:wetterstationen:ecowitt-stationen#die_ecowitt_iot_-_sensoren_iot-geraete

phifogg, 2025-01-17:
Hm. this seems more complex and I don't know if the sainlogic adapter would be the best place to add it to. So far the adapter is simpley reading (or receiving) data. Nothing is sent out to the station or any of the sensor.

I'd be happy to take a shot at it, but I do not have one of these which would make testing really hard. Is there a way I could access to it?

phifogg, 2026-01-21:
1.2.0-alpha.1 has first steps to allow controlling the WFC sensor.

Homeworker11, 2026-04-13:
Ja; ich kann das mit einem kleinen Blockly nachbilden.
Eleganter wäre es natürlich, wenn der Adapter es gleich mitliefern würde. Das macht es Übersichtlicher.