phifogg/ioBroker.sainlogic #226: Default value type !== value type

ticaki, 2025-01-22:
**Describe the bug**  
```
Object sainlogic.0.weather.current.solarradiation is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
```

vielleicht sowas in der Art beim Erstellen der States verwenden: (ist aus einer Rücksetzfunktion von mir)

```
                        let newVal: -1 | '' | '{}' | '[]' | false | null | undefined;
                        switch (state.stateTyp) {
                            case 'string':
                                if (typeof state.val == 'string') {
                                    if (state.val.startsWith('{') && state.val.endsWith('}')) {
                                        newVal = '{}';
                                    } else if (state.val.startsWith('[') && state.val.endsWith(']')) {
                                        newVal = '[]';
                                    } else {
                                        newVal = '';
                                    }
                                } else {
                                    newVal = '';
                                }
                                break;
                            case 'bigint':
                            case 'number':
                                newVal = -1;
                                break;

                            case 'boolean':
                                newVal = false;
                                break;
                            case 'symbol':
                            case 'object':
                            case 'function':
                                newVal = null;
                                break;
                            case 'undefined':
                                newVal = undefined;
                                break;
                        }
```

**Versions:**  
 - Adapter version: v0.11.6
 - JS-Controller version: 7.0.6
 - Node version: v20.18.1

mcm1957, 2025-01-22:
Viel zu viel Aufwand.
Eventuell ist da ein Fehler in io-package.json (wenn typ und default dort definiert werden)
Ansonsten extendState verwenden - das zieht das von selbst auf grade. Wenn die Paramater passen, ist exendObject ein NOP, sonst korrigeurt es sie.
Und wahrscheinlich ist da beim Objet ein Typ number definiert und der Defautl als Sting angegeben... Also einfach den Code korrigieren.

ticaki, 2025-01-22:
Ja man kann es sich auch schwer machen und alles einzeln per hand definieren.

```
<head></head>
2025-01-21 15:04:17.030 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.outdoortemp is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
--
2025-01-21 15:04:17.031 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.outdoorhumidity is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.031 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.dewpointtemp is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.031 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.windchilltemp is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.031 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.winddir is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.032 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.windspeed is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.032 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.windgustspeed is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.032 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.rain is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.033 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.dailyrain is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.033 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.weeklyrain is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.033 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.monthlyrain is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.033 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.yearlyrain is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.034 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.totalrainin is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.034 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.solarradiation is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.034 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.uvi is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.035 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.indoortemp is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.035 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.indoorhumidity is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.035 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.pressureabs is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.035 - warn: sainlogic.0 (167234) Object sainlogic.0.weather.current.pressurerel is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0!
2025-01-21 15:04:17.057 - info: sainlogic.0 (167234) State value to set for "sainlogic.0.weather.current.outdoortemp" has to be type "number" but received type "string"
2025-01-21 15:04:17.057 - info: sainlogic.0 (167234) State value to set for "sainlogic.0.weather.current.outdoorhumidity" has to be type "number" but received type "string"
2025-01-21 15:04:17.058 - info: sainlogic.0 (167234) State value to set for "sainlogic.0.weather.current.dewpointtemp" has to be type "number" but received type "string"
2025-01-21 15:04:17.058 - info: sainlogic.0 (167234) State value to set for "sainlogic.0.weather.current.windchilltemp" has to be type "number" but received type "string"
2025-01-21 15:04:17.058 - info: sainlogic.0 (167234) State value to set for "sainlogic.0.weather.current.winddir" has to be type "number" but received type "string"
[... cut ...]
2025-01-21 15:04:17.064 - info: sainlogic.0 (167234) State value to set for "sainlogic.0.weather.current.indoorhumidity" has to be type "number" but received type "string"
2025-01-21 15:04:17.064 - info: sainlogic.0 (167234) State value to set for "sainlogic.0.weather.current.pressureabs" has to be type "number" but received type "string"
2025-01-21 15:04:17.065 - info: sainlogic.0 (167234) State value to set for "sainlogic.0.weather.current.pressurerel" has to be type "number" but received type "string"

```

phifogg, 2025-01-22:
Kann es sein, dass Deine Installation schon älter ist? Ich bin der Meinung den Fehler bereits behoben zu haben. Dazu musst leider einmal alle State's löschen und vom Adapter neu anlegen lassen. Ich weiss leider keine elegante Lösung um das einmalig beim Start des Adapter mit zu machen.

phifogg, 2025-01-29:
Ok, danke. Das kann ich mir gerne mal anschauen.
Erklärt aber noch nicht warum Du die Meldungen bekommst.
Kann es sein, dass Deine Station irgendwelche komischen Werte schickt? Poste doch bitte mal die Daten die da kommen.

phifogg, 2025-01-29:
Oh.. du bist ja schnell :D 

GHenau das habe ich auch gerade gefunden. Ich hab nur die constants.js durchsucht weil ich eigentlich alles dort definiere. Aber eben nicht den default... schlechtes Design.

Ich habs geändert. Mal ausprobieren... die nächste Version sollte es beheben.

mcm1957, 2025-01-29:
Dann dass du dich der Sache so intensiv annimmst.

phifogg, 2025-01-29:
Kein Problem, ich bastel eh gerade an einem Enhancement Request....

ticaki, 2025-01-29:
Denke ich brauche dann die Werte meiner Station nicht zu posten :) 

Aus einer Fehlermeldung selbst verursacht von heute morgen:

`
/weatherstation/updateweatherstation.php?ID=xxx&PASSWORD=xxx&tempf=41.2&humidity=89&dewptf=38.1&windchillf=38.5&winddir=285&windspeedmph=4.25&windgustmph=5.37&rainin=0.071&dailyrainin=0.169&weeklyrainin=0.811&monthlyrainin=1.602&yearlyrainin=1.602&totalrainin=1.602&solarradiation=82.30&UV=0&indoortempf=77.5&indoorhumidity=33&absbaromin=28.378&baromin=29.840&lowbatt=0&dateutc=now&softwaretype=EasyWeatherPro_V5.1.8&action=updateraw&realtime=1&rtfreq=5

`