phifogg/ioBroker.sainlogic #227: LDS values are not available

hugo419752, 2025-02-04:
Hello all,

Unfortunately, the values from the LDS sensor are not displayed in the instance, could this be the case?

BR

Martin

phifogg, 2025-02-04:
Hi Martin,
what is an LDS sensor?
If you can run the adatper in debug mode I can look at the values it receives. Most likely it is a simple case of mapping the data zu IOBroker states.
Daniel

hugo419752, 2025-02-05:
Hi Daniel,

Ecowitt has a sensor available that's named LDS01 Laser Distance Sensor 

https://www.ecowitt.com/shop/goodsDetail/338

Since I don't know how the debug mode works, I can't provide you with that either.

Actually, only the three values need to be read out:

{"PASSKEY":"XXXXXXXXXXXXXXXXX","stationtype":"GW1100A_V2.3.7","runtime":"395108","heap":"23948","dateutc":"2025-02-05 07:28:23","tempinf":"76.10","humidityin":"33","baromrelin":"29.205","baromabsin":"29.205","rainratein":"0.000","eventrainin":"0.000","hourlyrainin":"0.000","dailyrainin":"0.000","weeklyrainin":"0.004","monthlyrainin":"0.004","yearlyrainin":"2.181","totalrainin":"2.181","soilmoisture1":"20","soilad1":"119","soilmoisture2":"0","soilad2":"102","soilmoisture3":"22","soilad3":"169","soilmoisture4":"0","soilad4":"105","soilmoisture5":"0","soilad5":"109","soilmoisture6":"32","soilad6":"192",**"air_ch1":"157","thi_ch1":"1480","depth_ch1":"1323"**,"wh40batt":"1.4","soilbatt1":"1.4","soilbatt2":"1.4","soilbatt3":"1.4","soilbatt4":"1.4","soilbatt5":"1.4","soilbatt6":"1.5","ldsbatt1":"3.14","freq":"868M","model":"GW1100A","interval":"60"}

BR

Martin

phifogg, 2025-02-05:
Hi Martin,
thanks, that already helps a ton.
Do you know if these attributes have some units? Any conversion needed like metric/imperial?
Daniel

hugo419752, 2025-02-05:
Hello Daniel,

the attributes have the following units

air_ch1 = mm (Metric)
thi_ch1 = mm (Metric)
depth_ch1 = mm (Metric)
ldsbatt1 = V (Volt)

Coversion should not be needed

BR 

Martin

closed 2025-02-06 by phifogg