phifogg/ioBroker.sainlogic #238: V0.12.2 kann nicht installiert werden

Horgul123, 2025-03-30:
Beim Versuch die neuste Version des Adpaters zu installieren kommt ein

`$ iobroker upgrade sainlogic@0.12.2 --debug

Update sainlogic from @0.12.1 to @0.12.2

NPM version: 10.8.2Installing iobroker.sainlogic@0.12.2... (System call)

npm error code ETARGET

npm error notarget No matching version found for admin@^7.4.10.

npm error notarget In most cases you or one of your dependencies are requestingnpm error notarget a package version that doesn't exist.

npm error A complete log of this run can be found in: /home/iobroker/.npm/_logs/2025-03-30T18_48_31_578Z-debug-0.log

Updating objects from io-package.json for adapter "sainlogic" with version "0.12.1"

upload [4] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/admin.d.ts admin.d.ts video/mp2t

upload [3] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/index_m.html index_m.html text/html

upload [2] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/sainlogic.png sainlogic.png image/png

upload [1] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/style.css style.css text/css

upload [0] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/words.js words.js application/javascript

Process exited with code 0`

Admin, js-controller usw. sind alle auf der neusten Beta Version

mcm1957, 2025-03-30:
Ja, die ist kaputt

Da haben nicht mal die Tests funktioniert. Aber offenbar wurde die Release manuell auf npm geschoben :-(

Siehe auch: https://github.com/ioBroker/ioBroker.repositories/pull/4672#issuecomment-2764460203

mcm1957, 2025-03-30:
@phifogg 
Ich habe mir erlaubt die defekte 0.12.2 aus dem Lates Repo zu nehmen damit nicht allzuviele Beta Tester da drüber fallen. 

In ca. 1/2h leifert das LATEST / BETA Repository wieder 0.12.1 aus. Sobald du eine gefixte 0.12.3 auf npm bereitstellst nimmt das repos dieser wieder automaisch mit. Ist also keine besondere Aktion deinerseits erforderlich (außer der Bau eine gexten Version :-) )

Horgul123, 2025-03-30:
@mcm1957 ah ok. Dank dir.  Mit dem Titel des anderen threads hatte ich kein Problem in Verbindung gebracht.

mcm1957, 2025-03-30:
@Horgul123 
mach das issue bitte wieder auf. wenn du kannst. Der link zeigt nur auf einen PR!. 

Das Problem muss noch bwhoben werdenbu dazu ist in issue sehr sinnvoll

Horgul123, 2025-03-31:
> [@Horgul123](https://github.com/Horgul123) mach das issue bitte wieder auf. wenn du kannst. Der link zeigt nur auf einen PR!.
> 
> Das Problem muss noch bwhoben werdenbu dazu ist in issue sehr sinnvoll

Ist erledigt :)

mcm1957, 2025-03-31:
@phifogg 
Ich glaub es reicht wenn du die admin Zeile auf package.json rauswirfst. Wenn was unklar ist meld dich ggF.

phifogg, 2025-03-31:
Ja danke. hab ich schon auf der 1.0.x Version gefixt. Muss noch ein backport machen...