phifogg/ioBroker.sainlogic #241: Update auf 1.0.12 schlägt fehl

Horgul123, 2025-04-24:
**Describe the bug**  
Beim Versuch auf 1.0.12 zu gehen kommt folgender Fehler:

`$ iobroker upgrade sainlogic@1.0.12 --debug

Update sainlogic from @0.12.1 to @1.0.12

NPM version: 10.8.2

Installing iobroker.sainlogic@1.0.12... (System call)

npm error code ETARGET

npm error notarget No matching version found for iobroker.sainlogic@1.0.12.

npm error notarget In most cases you or one of your dependencies are requestingnpm error notarget a package version that doesn't exist.npm error A complete log of this run can be found in: /home/iobroker/.npm/_logs/2025-04-24T08_00_47_647Z-debug-0.log

Updating objects from io-package.json for adapter "sainlogic" with version "0.12.1"

upload [4] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/admin.d.ts admin.d.ts video/mp2t

upload [3] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/index_m.html index_m.html text/html

upload [2] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/sainlogic.png sainlogic.png image/png

upload [1] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/style.css style.css text/css

upload [0] sainlogic.admin /opt/iobroker/node_modules/iobroker.sainlogic/admin/words.js words.js application/javascript

Process exited with code 0`

**Versions:**  
 - Adapter version: 1.0.12
 - JS-Controller version: 7.0.7
 - Node version: v20.19.0
 - Operating system: Debian

mcm1957, 2025-04-24:
Release 1.0.12 existiert auf npm nicht.
Möglicherweise wurde diese kurzzeitig deployed und wieder entfernt. 

Update des Repositories wurde veranlasst.

Horgul123, 2025-04-24:
@mcm1957 Dank Dir! Hat anscheinend auch die REST Api betroffen. Da hatte ich das auch und die wird jetzt auch nicht mehr als Update angeboten

closed 2025-04-24 by Horgul123