phifogg/ioBroker.sainlogic #248: Errors receiving data from Froggit WH3000

silmaril42, 2025-09-26:
I am trying to receive data from a Froggit WH3000 weather station. This should be the same as Ecowitt WS2910 and Sainlogic 3500, if this table is correct: https://meshka.eu/Ecowitt/dokuwiki/doku.php?id=start#a_brand_model_and_sensor_name_translation_table.

## Wunderground mode

When I set the listener to Wunderground mode, I need to set both Station ID and Station Key on the weather station to make it send requests.

The Path is set to `/weatherstation/updateweatherstation.php?` on both ends.

This leads to ioBroker warnings like this:
```
warn Listener received illegal request: /weatherstation/updateweatherstation.php?ID=A&PASSWORD=B&tempf=69.3&humidity=63&dewptf=56.1&windchillf=69.3&winddir=1&windspeedmph=0.00&windgustmph=0.00&rainin=0.000&dailyrainin=0.000&weeklyrainin=0.000&monthlyrainin=0.000&yearlyrainin=0.000&totalrainin=0.000&solarradiation=0.96&UV=0&indoortempf=69.4&indoorhumidity=60&absbaromin=28.201&baromin=29.929&temp2f=69.4&humidity2=61&temp3f=69.4&temp4f=70.2&temp5f=70.0&temp6f=69.3&humidity6=77&lowbatt=0&dateutc=now&softwaretype=EasyWeatherPro_V5.2.2&action=updateraw&realtime=1&rtfreq=5
```

Does this contain variable names the ioBroker adapter doesn't know?
Could you please add those?

Would it be possible to process all known variables and generate a warning for the unknown ones?

## Ecowitt mode

I also tested Ecowitt mode with Path `/data/report/`.

This works, but I got a bunch of warnings and infos of this kind:

```
warn Object sainlogic.0.weather.current.indoortemp is invalid: Default value has to be type "number" but received type "string" This will throw an error up from js-controller version 7.0.0! 
info State value to set for "sainlogic.0.weather.current.indoortemp" has to be type "number" but received type "string" 
```

This sounds like something that should be improved in the adapter to avoid getting errors in the future?

## Versions
 - Adapter version: v0.11.6
 - JS-Controller version: 7.0.7 
 - Node version: v20.19.5 
 - Operating system: Debian GNU/Linux 11 (bullseye) x86_64

The great news is that this works in Ecowitt mode :-)
I still hope for improvements of the things I described.

phifogg, 2025-09-27:
Hi @silmaril42 

thanks for the report.

The illegal request message is almost any time something wrong with the URL setup either on the weatherstation or in the adapter. I have seen some weatherstations where you need to remove the first slash (/) from the URL. Can you try that?

As for missing attributes, once the setup is correct, the adapter will log all attributes it doesn't know. We can add these quickly then.

Daniel

silmaril42, 2025-09-28:
Hi @phifogg,

thanks for your suggestion to play with the URL details!

I tried the URL with and without starting slash on both ends.
Without the slash in the weather station, nothing happens at all (no log entries, no data changes).
With the slash in the weather station, it doesn't matter if I have it in iobroker, I always get an illegal request warning.

But I did find a combination that works for me:
- iobroker: `/weatherstation/updateweatherstation.php`
- weather station: `/weatherstation/updateweatherstation.php?`

It seems the question mark at the end of the URL cannot be present in iobroker, but has to be there in the weather station.

This looks like a minor issue in the way the incoming request path is parsed, that could be improved to handle things correctly both with and without this question mark.

With this setup, I also noticed a data inconsistency between Wunderground and Ecowitt modes.
In Ecowitt mode I get current values for additional sensors 1 to 5.
In Wunderground mode I get the same values as 2 to 6.

I guess this is not the adapter's fault, but the weather station is sending different numbers in different modes.
This might be confusing if you don't know it, but I wouldn't recommend doing anything "intelligent" here on the adapter side. Better keep this as it is.